orocos/soem #4: Can't find osal.h when linking from catkin/ROS package

shaun-edwards, 2015-07-11:
System config:
 - ROS Indigo/14.04
 - Installed `ros-indigo-soem`

When trying to link against `soem` to build the [robotiq](https://github.com/shaun-edwards/robotiq/tree/issue39_soem_lib) package, I get the following error in `ethercattype.h`

```
/opt/ros/indigo/include/soem/ethercattype.h:68:18: fatal error: osal.h: No such file or directory
 #include <osal.h>
```

If I download the soem source, the header is found and everything compiles.

I can also get it to work by manually adding the soem header directory, see [here](https://github.com/shaun-edwards/robotiq/blob/issue39_soem_lib/robotiq_ethercat/CMakeLists.txt#L18)

I'm not sure what the fix is, since catkin/ROS convention is for headers to be in the `<package_name>` subdirectory, but the soem headers do not adhere to this.

smits, 2015-07-13:
This should have been fixed by 40a0aa07d69, can you check your version of ros-indigo-soem. I have the osal.h header in /opt/ros/indigo/include/soem. I do not understand how it could find the ethercattype.h header in the exact same path, but not the osal.h header.

shaun-edwards, 2015-07-13:
I have the latest version.  I have verified that the headers are installed correctly.  It can find `ethercattype.h` because I pre-pend the the `soem` package to the include, [here](https://github.com/shaun-edwards/robotiq/blob/issue39_soem_lib/robotiq_ethercat/src/ethercat_manager.cpp#L10).  By ROS convention, all includes should be `<package_name>/<path>`

The error occurs because `ethercattype.h` does not "follow" this convention, see [here](https://github.com/smits/soem/blob/master/soem/ethercattype.h#L68)

These conventions are just that, and pretty much Cmake magic to me.

shaun-edwards, 2015-07-13:
I'm curious why others aren't having this issue?

smits, 2015-07-14:
I know we are violating some of the ROS conventions, by not prefixing the
headers with soem (I wanted to avoid adapting to much soem code) but we
explicitely export the the soem folder as an include directory here:
https://github.com/smits/soem/blob/master/CMakeLists.txt#L7 in case you are
including soem as a catkin package in source and here
https://github.com/smits/soem/blob/master/soem-config.cmake.in#L8 in case
you are building against an installed version.

So it seems you are somehow missing the
include_directories(${catkin_INCLUDE_DIRS}) or the
include_directories(${soem_INCLUDE_DIRS})

smits, 2015-07-14:
Could you try to replace the find_package(catkin ... soem) by find_package(soem), maybe we messed something up with the generated cmake config files.

shaun-edwards, 2015-07-15:
I would hope that releasing in ROS under Catkin wouldn't require you to change your directory layout.  I tried replacing `catkin...` with `find_package(soem)`, but it did not solve the problem.

I printed out the following Cmake `<package>_INCLUDE_DIRS` variables to try to debug the problem.

```
SOEM Include directories: /opt/ros/indigo/include
Catkin Include directories: /opt/ros/indigo/include/usr/include
Robotiq Ethercat Include directories: 
```

While the `osal.h` header is installed:
```
ls /opt/ros/indigo/include/soem -l
total 104
-rw-r--r-- 1 root root  5242 Jan 26 15:25 ethercatbase.h
-rw-r--r-- 1 root root  5521 Jan 26 15:25 ethercatcoe.h
-rw-r--r-- 1 root root  2784 Jan 26 15:25 ethercatconfig.h
-rw-r--r-- 1 root root  7526 Jan 26 15:25 ethercatconfiglist.h
-rw-r--r-- 1 root root  2635 Jan 26 15:25 ethercatdc.h
-rw-r--r-- 1 root root  2676 Jan 26 15:25 ethercatfoe.h
-rw-r--r-- 1 root root 18584 Jan 26 15:25 ethercatmain.h
-rw-r--r-- 1 root root  2376 Jan 26 15:25 ethercatprint.h
-rw-r--r-- 1 root root  5349 Jan 26 15:25 ethercatsoe.h
-rw-r--r-- 1 root root 17586 Jan 26 15:25 ethercattype.h
-rw-r--r-- 1 root root  4923 Jan 26 15:25 nicdrv.h
-rw-r--r-- 1 root root  2531 Jan 26 15:25 osal.h
-rw-r--r-- 1 root root  1580 Jan 26 15:25 oshw.h
```
The includes above do not provide the appropriate root directory.

shaun-edwards, 2015-07-15:
I found a work-around for this, see [here](https://github.com/shaun-edwards/robotiq/commit/13bc1ad0735d0dbdd3f7a98f98168c40322caf2b).  I'm not sure how/if this will work on the build servers, but we will see.

smits, 2015-07-15:
Wow, your catkin include dirs look completely wrong, I'll try to reproduce now, I assume you're using Indigo?

shaun-edwards, 2015-07-15:
Yes. Indigo/14.04.

wujinghust, 2016-02-26:
I input   source ./setup.sh linux   in the terminal,it is normal. Later, I input    make all   in the  terminal   there are serveral error as fellow:

-- Compiling ebox.c
ebox.c: In function ‘eboxtest’:
ebox.c:196:26: warning: format ‘%lld’ expects argument of type ‘long long int’, but argument 3 has type ‘int64’ [-Wformat]
ebox.c: In function ‘ecatthread’:
ebox.c:284:8: warning: variable ‘rc’ set but not used [-Wunused-but-set-variable]
ebox.c: In function ‘main’:
ebox.c:348:8: warning: variable ‘iret1’ set but not used [-Wunused-but-set-variable]
--- Linking ebox
/usr/bin/ld: cannot find -lsoem
collect2: ld returned 1 exit status
make[3]: *** [ebox] Error 1
make[2]: *** [subdirs] Error 2
make[1]: *** [linux] Error 2
make: *** [test] Error 2

Timple, 2018-02-13:
> I'm curious why others aren't having this issue?

We are having the issues as well under kinetic.

jproberge, 2018-07-24:
> I'm curious why others aren't having this issue?

Same here, having this exact problem under kinetic as well

mgruhler, 2018-11-29:
How to do this is now documented in the [README.md](https://github.com/mgruhler/soem#Installation).
Please check the section [Development](https://github.com/mgruhler/soem#Development) if you want to use this package from source.

closed 2018-11-29 by mgruhler