orocos/soem #5: Release for ROS Jade?

davetcoleman, 2016-02-18:
Package robotiq_ethercat depends on this package so cannot currently be installed on Jade from debian. Could someone release this package again? Thanks!

shaun-edwards, 2016-12-22:
+1...We have some large changes in PR ros-industrial/robotiq#63 that I would like to pull into the `jade` development branch.  The lack of a jade version of soem is holding things up.  I'm happy to help out if needed.

shaun-edwards, 2017-01-17:
Friendly ping.

smits, 2017-01-17:
Would you mind if we skip Jade altogether and directly release kinetic?

R.

shaun-edwards, 2017-01-18:
Jade is about to EOLed...Kinetic is definitely the priority.

k-okada, 2017-06-23:
hello, sorry to bother you, but can I ask you to release on kinetic?

If you think you do not have enough resource to maintain this package, could you consider give @k-okada and @7675t for write access to both this package and https://github.com/smits/soem-gbp
If you think you could not keep this package in  your repository, Could you transfer these repository to https://github.com/ros-drivers and https://github.com/ros-gbp ? or can we host this repositories?

As I discussed in https://discourse.ros.org/t/releasing-repositories-form-other-people/1797/4 , I understand keep maintain the software is not a easy thing, and also it is not easy for others to take over the maintainership, because we, at least me, do not fully understand the source code of this package so it is hard for me to judege if incoming PR is correct or not, or cope with bug reportes and feture requests, but we still willing to use this software in new ROS distribution such as kinect.
What I proposed there is to create a group of maintainers (https://github.com/smits/soem/pull/10) who mainly responsible for releasing jobs, so that at least we can continue using the software in future.

v4hn, 2017-06-29:
We are waiting for a kinetic release over here as well. +1

smits, 2017-06-29:
I should be able to do this quickly. If all goes well there should be a
release pr on rosdistro tomorrow.

R.

smits, 2017-06-30:
@shaun-edwards @k-okada See https://github.com/ros/rosdistro/pull/15414 I'm fine with moving soem and soem-gbp to ros-drivers and ros-gbp, I wonder if it would be easier to release directly the upstream https://github.com/OpenEtherCATsociety/SOEM instead of keeping this fork.

davetcoleman, 2017-06-30:
What is the difference between this SOEM and the upstream version?

k-okada, 2017-07-02:
smits/soem has `$Id: osal.c 416 2013-01-08 21:54:25Z smf.arthur $` (https://github.com/smits/soem/blob/master/osal/linux/osal.c#L17)
and OpenEtherCATsociety/SOEM 's inital commit was `$Id: osal.c 464 2013-03-13 20:40:25Z smf.arthur $` (
https://github.com/OpenEtherCATsociety/SOEM/blob/7472dae1a275b103283147b923ea0c242a3e92fe/osal/linux/osal.c#L17)
so it should be between 1.2.8 and 1.3.0 https://github.com/OpenEtherCATsociety/SOEM/blob/7472dae1a275b103283147b923ea0c242a3e92fe/ChangeLog#L51-L80

At this moment, OpenEtherCATsociety have version v1.3.1 https://github.com/OpenEtherCATsociety/SOEM/blob/v1.3.1/osal/linux/osal.c and 
we have more than 60 commit since then 
https://github.com/OpenEtherCATsociety/SOEM/compare/v1.3.1...master

Did someone using upstream version?

k-okada, 2017-07-02:
@smits thank you for your efforts, it seems you need to answer to https://github.com/ros/rosdistro/pull/15414#discussion_r125042070 to get them releasing.

MatthijsBurgh, 2017-07-02:
The differences between 1.3.0 and 1.3.1 are shown here: http://openethercatsociety.github.io/doc/soem/index.html

As mentioned in issue #9 , I tried using 1.3.1 myself. It did compile, but wasn't working at all. I think the main issue is the ```Added multi-threaded configuration for parallel configurations of slaves```

mgruhler, 2018-11-30:
as kinetic has been released, I'll close this one.

closed 2018-11-30 by mgruhler