orocos/soem #7: compile the soem library

wujinghust, 2016-02-26:
when I make the makefile in soem directory, there is a error as fellow:

make: *** No rule to make target `simple_test.o', needed by `simple_test'.  Stop.

what should I do ? thank for you help。

smits, 2016-02-26:
Depending on what you exactly want to do you could also take a look at the
official repos and see if it is fixed there:
https://github.com/OpenEtherCATsociety/SOEM
This repos is a fork that is typically build as part of a catkin workspace
and uses cmake instead of the original instructions.

wujinghust, 2016-02-27:
thanks for you help ! 
I copy oshw.c oshw.h osal.h osal.c nicdrv.c nicdrv.h to the soem directory, and I modify the makefile.
but there are errors when I compile the makefile .

make: Circular simple_test.o <- simple_test.o dependency dropped.
gcc -O2 -lpthread -lm -lrt -lc -lpq -Wall -o ethercatconfig.o ethercatconfig.c ethercatconfig.h ethercatcoe.c ethercatcoe.h ethercattype.h osal.c osal.h oshw.h oshw.c ethercatmain.c ethercatmain.h ethercatbase.c ethercatbase.h ethercatsoe.h ethercatsoe.c
In file included from oshw.h:29:0,
                 from ethercatconfig.c:52:
ethercattype.h:68:18: fatal error: osal.h: No such file or directory
compilation terminated.
ethercatconfig.h:66:21: error: unknown type name ‘ecx_contextt’
ethercatconfig.h:66:44: error: unknown type name ‘uint8’
ethercatconfig.h:67:26: error: unknown type name ‘ecx_contextt’
ethercatconfig.h:67:63: error: unknown type name ‘uint8’
ethercatconfig.h:68:23: error: unknown type name ‘ecx_contextt’
ethercatconfig.h:68:46: error: unknown type name ‘uint16’
ethercatconfig.h:69:24: error: unknown type name ‘ecx_contextt’
ethercatconfig.h:69:47: error: unknown type name ‘uint16’
In file included from oshw.h:29:0,
                 from ethercatcoe.c:51:
ethercattype.h:68:18: fatal error: osal.h: No such file or directory
compilation terminated.
ethercatcoe.h:64:4: error: unknown type name ‘uint16’
ethercatcoe.h:66:4: error: unknown type name ‘uint16’
ethercatcoe.h:68:4: error: unknown type name ‘uint16’
ethercatcoe.h:70:4: error: unknown type name ‘uint16’
ethercatcoe.h:72:4: error: unknown type name ‘uint8’
ethercatcoe.h:74:4: error: unknown type name ‘uint8’
ethercatcoe.h:76:31: error: ‘EC_MAXNAME’ undeclared here (not in a function)
ethercatcoe.h:83:4: error: unknown type name ‘uint16’
ethercatcoe.h:85:4: error: unknown type name ‘uint8’
ethercatcoe.h:87:4: error: unknown type name ‘uint16’
ethercatcoe.h:89:4: error: unknown type name ‘uint16’
ethercatcoe.h:91:4: error: unknown type name ‘uint16’
ethercatcoe.h:112:19: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:112:42: error: unknown type name ‘uint16’
ethercatcoe.h:112:56: error: unknown type name ‘uint16’
ethercatcoe.h:112:70: error: unknown type name ‘uint8’
ethercatcoe.h:112:84: error: unknown type name ‘int32’
ethercatcoe.h:113:17: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:113:40: error: unknown type name ‘uint16’
ethercatcoe.h:113:54: error: unknown type name ‘uint16’
ethercatcoe.h:113:68: error: unknown type name ‘uint8’
ethercatcoe.h:114:23: error: unknown type name ‘boolean’
ethercatcoe.h:115:18: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:115:41: error: unknown type name ‘uint16’
ethercatcoe.h:115:55: error: unknown type name ‘uint16’
ethercatcoe.h:115:69: error: unknown type name ‘uint8’
ethercatcoe.h:116:5: error: unknown type name ‘boolean’
ethercatcoe.h:117:15: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:117:38: error: unknown type name ‘uint16’
ethercatcoe.h:117:52: error: unknown type name ‘uint16’
ethercatcoe.h:118:15: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:118:38: error: unknown type name ‘uint16’
ethercatcoe.h:118:52: error: unknown type name ‘uint16’
ethercatcoe.h:119:20: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:119:43: error: unknown type name ‘uint16’
ethercatcoe.h:120:22: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:120:45: error: unknown type name ‘uint16’
ethercatcoe.h:121:20: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:121:43: error: unknown type name ‘uint16’
ethercatcoe.h:122:27: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:122:50: error: unknown type name ‘uint16’
ethercatcoe.h:123:22: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:123:45: error: unknown type name ‘uint16’
ethercatcoe.h:123:58: error: unknown type name ‘uint8’
ethercatcoe.h:124:16: error: unknown type name ‘ecx_contextt’
ethercatcoe.h:124:39: error: unknown type name ‘uint16’
ethercattype.h:68:18: fatal error: osal.h: No such file or directory
compilation terminated.
[... 242 lines cut ...]
ethercatsoe.h:71:16: note: originally defined here
ethercatsoe.h:86:4: error: unknown type name ‘uint16’
ethercatsoe.h:97:1: error: unknown type name ‘PACKED_END’
ethercatsoe.h:101:1: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘typedef’
ethercatsoe.h:101:16: error: redefinition of ‘struct PACKED’
ethercatsoe.h:83:16: note: originally defined here
ethercatsoe.h:104:4: error: unknown type name ‘uint16’
ethercatsoe.h:109:1: error: unknown type name ‘PACKED_END’
ethercatsoe.h:126:1: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘typedef’
ethercatsoe.h:126:16: error: redefinition of ‘struct PACKED’
ethercatsoe.h:101:16: note: originally defined here
ethercatsoe.h:129:4: error: unknown type name ‘uint32’
ethercatsoe.h:157:1: error: expected ‘=’, ‘,’, ‘;’, ‘asm’ or ‘__attribute__’ before ‘int’
ethercatsoe.h:158:18: error: unknown type name ‘ecx_contextt’
ethercatsoe.h:158:41: error: unknown type name ‘uint16’
ethercatsoe.h:158:55: error: unknown type name ‘uint8’
ethercatsoe.h:158:70: error: unknown type name ‘uint8’
ethercatsoe.h:158:90: error: unknown type name ‘uint16’
ethercatsoe.h:159:20: error: unknown type name ‘ecx_contextt’
ethercatsoe.h:159:43: error: unknown type name ‘uint16’
In file included from oshw.h:29:0,
                 from ethercatsoe.c:50:
ethercattype.h:68:18: fatal error: osal.h: No such file or directory
compilation terminated.
make: *** [ethercatconfig.o] Error 1

mgruhler, 2018-11-29:
Readme updated, or rather, a new one added, clarifying thins for ROS...
I consider this fixed :-)

closed 2018-11-29 by mgruhler