orocos/soem #8: Using soem with robotiq

jontoto, 2017-02-01:
I'm trying to build the soem library, but when linking with the robotiq packages I run into an issue where cmake cannot find the soem/ethercattype.h file. It seems like there have been other similar issues with robotiq packages. I'm running on kinetic, is there a way to get them to build together?

keerthanamanivannan, 2017-03-29:
Did you figure out a way to fix this?

newuhe, 2017-12-22:
I have the same problem, have you fixed it ???

mgruhler, 2018-11-29:
How to do this is now documented in the [README.md](https://github.com/mgruhler/soem#Installation).
Please check the section [Development](https://github.com/mgruhler/soem#Development) if you want to use this package from source.

closed 2018-11-29 by mgruhler