orocos/soem #19: ROS SOEM installation

mictas, 2019-06-17:
Hi,
I'm new about SOEM and I want to test it using ROS kinetic in Ubuntu 16.04.
I followed the instructions regarding installation which I found at the link: https://github.com/mgruhler/soem
Unfortunately, installation doesn't work and the following errors are shown:
Errore:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-soem amd64 1.3.0-0xenial-20180809-141333-0800
  404  Not Found 
E: Impossibile recuperare http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-soem/ros-kinetic-soem_1.3.0-0xenial-20180809-141333-0800_amd64.deb  404  
Does it exist another link to download it?
How can I solve this issue?
Thank you.

mgruhler, 2019-06-18:
SOEM ist released into kinetic, thus the above should work. However, you might have to [Update your gpg Keys](https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454) for apt to work again.

For using a source Checkout, please Check the [readme](https://github.com/mgruhler/soem/blob/master/README.md#Usage) and report more Details in Errors.

mictas, 2019-06-18:
Hi, I followed your suggests and after the update of gpg keys, installation worked correctly. Thank you.

closed 2019-06-18 by mictas