orocos/soem #20: SOEM for ROS 2.0

mictas, 2019-06-18:
Hi,

is a future release of SOEM for ROS 2 (Crystal or Dashing Diademata) expected?
I'm getting familiar with ROS 2 and I think that a SOEM release for it it would be useful.
Thank you.

mgruhler, 2019-06-18:
This would be beneficial, I agree.
But I'm not familiar with the ROS2 build system as of now and don't think I'll be able to invest time on this in the near future.

Thus, I cannot judge what effort a "port" to colcon would be. I'm assuming this would require a separate branch with separate build instructions, doubling maintenance effort.

That being said: if someone is willing to invest the time in a colcon port, I'll be happy to investigate any outcomes and track/support this in this repo. Also, I'll be happy to share any maintenance for this repository, if someone is willing to invest his time.

This, please reply to this issue if you want to start working on a colcon port.

mictas, 2019-06-18:
I understand. Currently I'm not able to develop a SOEM ROS2 package and I won't have the possibility to do it in the future. I hope that someone else could develop it soon. Thak you.

JensLepers, 2019-12-13:
Hi, 
I'm also interested in using SOEM in a ROS2 environment, is there already an update ?
I'm also not able to develop it all by myself but I am available to help investigate this problem.
Please let me know what the future plans are with this project

mgruhler, 2019-12-13:
@JensLepers I don't have any ROS2 setup, so I cannot do this at this point of time (and also probably not in the next few months). Again, any help here is appreciated...

jLepers, 2019-12-13:
I have a ROS2 setup and I am wondering how much work it would be to make the change. Can I assume that I only need to change the CMakeList? (use colcon to build the packages). Or am I completely underestimating this ?

mgruhler, 2020-01-10:
@jLepers, sorry for the late reply (christmas holdays, etc...). Happy new year.

As this is a very simple wrapper, in theory only the `package.xml` and the `CMakeLists.txt` would need to be touched.

Checking the [migration guide](https://index.ros.org/doc/ros2/Contributing/Migration-Guide) actually says for [the `package.xml`](https://index.ros.org/doc/ros2/Contributing/Migration-Guide/#package-manifests), that this should be fine already. As it is already format 2 and doesn't have any dependencies (except for the  `buildtool_depend` on `catkin`, which needs to be changed, see next paragraph) this should be okay.

For [the build system / `CMakeLists.txt` migration](https://index.ros.org/doc/ros2/Contributing/Migration-Guide/#build-system) the relevant steps should be documented. It is important to note that the [`cmake/soem-ros-extras.cmake.in`]() file would need to be adapted as well.
And then there is the obvious change in the `package.xml`

```xml
<!--  <buildtool_depend>catkin</buildtool_depend> -->
  <buildtool_depend>ament_cmake</buildtool_depend>
```

I'd be happy to except PRs to a ROS2 branch, but as long as I cannot test is, the work would need to be done by someone else.

edhage, 2020-05-01:
I made a ROS2 wrapper for SOEM and I can share it if you want. Actually think about making a another package which is using the SOEM-package it as an example of how to use it in ROS2 with some beckhoff I/O, and sharing it publicly on bitbucket so you can take it from there.  Got something working now in a ROS2 C++node where the connection is nicely terminated in the destructor of the node. I made one general header-file that is to be used in the application-package so it only needs one include: "soem/soem.h"

mgruhler, 2020-05-01:
@edhage Sharing what you have would be great.

I'm not sure what you suggest, though. Do you want to add it to this package or create a new one? Any examples would be great, though.

edhage, 2020-05-01:
@mgruhler  So I made two repositories. https://bitbucket.org/edhage/soem_ros2  for the ROS2 package. Please look at it, and integrate it in your package if you want. If that is done I could remove this repository (no point in keeping doubles).

Another repository is a working example, https://bitbucket.org/edhage/ether_ros2, that uses soem_ros2. 

So if I put both in a workspace and build ("colcon build") than both packages build in the right order (soem_ros2 first) and I can use it directly.

cplasberg, 2020-07-30:
this is a forked working version for Foxy on 20.04 with just some changes to CMakeLists.txt and package.xml
[https://github.com/cplasberg/soem/tree/foxy-devel](https://github.com/cplasberg/soem/tree/foxy-devel).

mgruhler, 2020-07-31:
@cplasberg thanks. This looks nice. I still haven't gotten around to set up a ROS2 system. But if this is working, I'd be happy to integrate this.
However, I'd need support with testing until I get a ROS2 system, which, to be honest, might be quite a while...
Would you be willing to support here?

cplasberg, 2020-09-17:
Sorry for my late response. I think I could do that.

gborghesan, 2022-03-08:
Also see https://github.com/gborghesan/soem
branch foxy-devel