orocos/soem #21: SOEM PDO Transmission

mictas, 2019-06-28:
Hi,
I've installed SOEM and I'm using it with ROS Kinetic on Ubuntu 16.04.
I've created a C++ ROS node and inside it I've reproduced the example "simple test" contained in the SOEM package to communicate to a DELTA servo drive.
The communication link between ROS node and servo started correctly, in fact I see the change of the state of the slave after each specific command.
When I try to launch "ec_send_processdata()", the servo drive give me an alarm message in which it told me that it is in operational state but the data receveid are wrong.
Servo drive configuration needs to receive a RxPDO containing 2 object ( a uint16 "control word" 6040 and a int32 "target position" 607A) but I don't understand how to do to send them to the slave.
How can I solve it?
Which variables must be setted?
Waiting for you response, thank you.

mgruhler, 2019-07-02:
Please add more details:
- Which drive are you using exactly?
- Have you configured the PDOs on the device?
- Have you checked out the tutorial [on the SOEM homepage](https://web.archive.org/web/20180217041851/https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html) (note that this is a link to the wayback machine, as there has been a [new release](https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html) for SOEM recently, and I didn't have the time to check what changed. Seems like a lot)?

In the end, you have to use the `ec_slave[X].outputs/inputs` to set and retreive data using PDOs. But how to configure them depends completely on your machine.

So please follow up with a more detailed description of what you are trying to do and make sure you understand the concepts of CoE and CanOpen of your device.

mictas, 2019-07-12:
Hi,
thank you for your answer, I followed your suggests (using the ec_slave[X].outputs/inputs) and the PDO communication between master and slave works correctly now. Are there any way to set the speed or frequency of data transmission? We are searching in the code but we haven't found anything yet.

mgruhler, 2019-07-12:
Well, there is a [`for`-loop](https://github.com/mgruhler/soem/blob/master/test/linux/simple_test/simple_test.c#L98-L122)  which does the actual communication during "operation", and a [`usleep()`](https://github.com/mgruhler/soem/blob/master/test/linux/simple_test/simple_test.c#L120) therein...
Does this help?

mictas, 2019-07-30:
Hi,
thank you for your answer, we have changed usleep() as you suggested and it works.
We need more clarifications about the ethercat communication between master and slaves.
We suppose to have N slaves connected to the master.
For each slave i (i=1,...,N) we can set values for RXPDO in ec_salve[i].outputs and read TXPDO from ec_slave[i].inputs. 
1) Reading SOEM files, we deduced that using "ec_send_processdata()" the master sends RXPDOs to all slave at the same time, while using "ec_receive_processdata(EC_TIMEOUTRET)" it receives TXPDOs from all slaves at the same time. During these operations SOEM library packs and unpacks these informations from the ethercat datagram independently. Have we understood right?
2) With the aim to establish an ethercat communication between master and slaves, using the "simple_test" example, we only need to set "ec_salve[i].outputs" and then send/receive data using "ec_send_processdata" and "ec_receive_processdata". I think it is not necessary to modify SOEM files. Am I wrong?
3) In "simple_test" code, at lines 109 and 115 are printed respectively "O" and "I" regarding "ec_slave[0]", which should be referred to the master. Why not the values regarding the slaves "ec_slave[i]"? Is there a specific reason for this? It is not clear for us.
Thank you.

mgruhler, 2019-08-02:
@mictas first a disclaimer: please note that this is just a wrapper for the [OpenEtherCATsociety/SOEM upstream repository](https://github.com/OpenEtherCATsociety/SOEM) and provides only a way to easily include this in ROS and package it for ROS in a debian package. So all detailed questions that are more about Ethercat in general can be found in the relevant literature and all questions targetting the SOEM source code specifically should be targeting the upstream repo. Note that with #23 being merged end of august, this repo will actually completely track the source code of the latest upstream release!

To your questions: I'm by no means an expert for Ethercat, but I'll try to do my best to help you solving your issues here, as long as it is not beyond the scope of this repo.

1. yes
1. yes, you don't need to modify any files in here. SOEM is a library for providing the ethercat interface and you only need to use the provided API in your one code / ROS node.
1. Well, this is a `simple_test`. So it is expected to work whatever slave you connect. As this can be basically anything, and you don't know how the slaves are configured, this is simply printing whatever is on the inputs and outputs of the master. What data would you want to send to a slave you don't know what it has configured?

mgruhler, 2019-08-02:
@mictas please close when this issue is resolved.

mictas, 2019-08-02:
Thank you very much for your answers. We cleared a lot of doubts. Best regards.

closed 2019-08-02 by mictas