orocos/soem #22: Issue when exporting a new package with SOEM

wongkaiweng, 2019-06-28:
Hi, 

I am trying to set up a new package (`pkg_a`) that depends on SOEM. I got that working properly. The problem appears when I use this new package with another package (`pkg_b`), I have to add `${soem_INCLUDE_DIRS}/soem` in the `include_directories` of `pkg_b` again. 

I am looking to find a solution that this is taken care of in `pkg_a`. Since I cannot add `${soem_INCLUDE_DIRS}/soem` in `catkin_package INCLUDE_DIRS` of `pkg_a` (I would get a non-package include path error from `catkin_lint`), I am curious if there is a way to do this. 

Thanks!

mgruhler, 2019-07-02:
Actually, exporting `${soem_INCLUDE_DIRS}/soem` in `pkg_a` is the way that I was using it so far as well.

If your problem is "only" automated tests, you [could "silence" `catkin_lint` by using the `--ignore external_include_path` flag](http://fkie.github.io/catkin_lint/#diagnostic-levels). Though I'm not sure you'd want to do that, depending on your test setup.

What should work is to include `soem` not as a catkin package, but as a system package and export that in `pkg_a`. I.e. use the following snippets instead of the typical `catkin`-way to do things:

```CMake
find_package(catkin REQUIRED ...)
find_package(soem REQUIRED)

catkin_package(
  ...
  DEPENDS
    soem
  ...
)

include_directories(
  ...
  ${soem_INCLUDE_DIRS}
  ${soem_INCLUDE_DIRS}/soem
)

target_link_libraries(${PROJECT_NAME} ${catkin_LIBRARIES} ${soem_LIBRARIES})
```
and then only `find_package( catkin REQUIRED COMPONENTS pkg_a)` in `pkg_b`.
A quick check in my repos (if I understood correctly, I am doing the same thing you do) seems to confirm this. If this does not work for you, please report back with a [SSCCE](http://www.sscce.org/).

Other than that, I'd like to link to #17 which when (if) a proper solution is available would also solve your issue.

wongkaiweng, 2019-07-02:
Thank you for the suggestion. I tried it out and here are my observations:

1. `find_package(soem REQUIRED)` is problematic. `catkin_make` can pass the configuration step but it cannot find references to functions. Here's is an example of the errors: `undefined reference to 'ec_SDOwrite'`. However, if `soem` is in `find_package(catkin REQUIRED COMPONENTS soem)` for example, then it compiles without an issue. 

2.  Putting `soem` under `DEPENDS` in `catkin_package` didn't work for me and here's the error from `catkin_lint`: `catkin_package() lists 'soem' as system package but it is not`. I would like to avoid 'silencing' `catkin_lint` if possible. This is related to 1. probably.  

I got the ROS version of SOEM. Could that be the reason why it is not a system package? I'm not sure how this is handled behind the scene. I will follow #17 too.

mgruhler, 2019-07-03:
about 1: This seems to be a linker error. Did you add `${soem_LIBRARIES}` to any `target_link_libraries` calls? This is all assuming you are actually using this in `pkg_a`.
If you call a `soem` function in `pkg_b` directly, you do have a direct dependency on `soem` and should thus add it to your `pkg_b` explicitly.
If both of the above are not the case, please describe your problem in more details with relevant code and a reproducible example.

about 2: I'm not sure how `catkin_lint` is actually deducing that `soem` is a catkin package in this case.  I was assuming that it is taking all `find_package'd packages, but this seems to not be the case. However, I can reproduce your problem. And I don't have any immediate solution.

This package is a wrapped version of SOEM for the use with ROS. You could also try the original SOEM (linked in the description and the README), but will probably run into a lot of the same issues you are having here :-)
In the end, I hope to find a solution by fixing #17 and #16, but I still don't have a roadmap or even an estimation when and how this might be done.

As a last resort, I just learned that you can also ignore `catkin_lint` errors in a single package or even only for a single line by adding the pragmas `#catkin_lint: ignore <ID>` at the top of the file or `#catkin_lint: ignore_once <ID>` directly before the statement, as described [here](http://fkie.github.io/catkin_lint/messages/).

mgruhler, 2019-07-05:
@wongkaiweng I just sent #23 which (among others) includes hopefully a solution to your problem. Actually, it is not even required/possible to specify `${soem_INCLUDE_DIRS}/soem` in your `pkg_a` anymore.
It would be great if you could check the PR out and leave a comment.

mgruhler, 2019-07-15:
#24 contains a similar solution.

PR #24 (2019-09-02): Upstream soem release 1.4.0 via subtree
This PR is meant to replace #23, if the decision to use subtree over submodules is made.

It also fixes #16, fixes #17 and fixes #22.

~Also, this is now a proof-of-concept for review and tests by other contributors. The commit history should definitely be cleaned...~ done

~The same issue holds as for in #23: this will be a breaking change for any consumers, since the previous work-around to the issue with header includes does not work anymore!~ found backwards compatible way

again, any feedback about this is highly welcome
closed 2019-09-02 by mgruhler