orocos/soem #26: No socket connection

wangyujin100, 2019-07-23:
Hi,
I've installed SOEM(build from source) with ROS Kinetic on Ubuntu 16.04.
And I've created a C++ ROS node and inside it I've reproduced the example "simple test"
However, when I "rosrun" this node, I was told that:
**No socket connection on ethxxx
Excecute as root**
Thus, I try "rosrun" this node in root:
**sudo su
rosrun [package] [node]**
but the same problem occurred.

How can I solve it?
Could you give some examples, such as a node?
Waiting for you response, thank you.

mgruhler, 2019-07-23:
It is expacted behavior that you need to run anything with SOEM as root. To quote the [SOEM docs "Getting Started" section](https://openethercatsociety.github.io/doc/soem/):

> First try (assume EtherCAT on eth0): sudo ./simple_test eth0 As SOEM uses RAW sockets it will need to run as root.

But I can also reproduce your behavior. This error message also pops up when you specify a non-existing ethernet device. Please make sure that the ethernet device you pass to your `simple_test` node does actually exist. I assume you changed the error log, as `ethxxx` is definitely not an existing interface. If not, this is already your problem.

If you later want to use SOEM in a ROS node, you should use [ethercat-grant](https://index.ros.org/r/ethercat_grant/#kinetic) utility.

As this issue seems to be with specifying the correct ethernet device, I'll close it. Please feel free to reopen if this does not solve your problems, but provide more details (i.e. an exact c'n'p of the error, as well as the output of `ifconfig`).

closed 2019-07-23 by mgruhler

wangyujin100, 2019-07-23:
Thank you for your reply. It's great helpful.