orocos/soem #27: Getting error while integrating with ros node in C++

MRIAC, 2019-08-02:
@mgruhler i have tried a lot to integrate soem library to our own ROS C++ node which will publish the topic... but i am unable to do, i have pointed my issue below. could you please help me out from this issue.if you need i can share my CMakeLists.txt file.

Errors     << ethercat_soem:make /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/logs/ethercat_soem/build.make.000.log
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp: In member function ‘void Epos_interaction::initialized_thread()’:
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp:595:73: warning: converting from ‘void (Epos_interaction::*)()’ to ‘void*’ [-Wpmf-conversions]
          osal_thread_create(&thread1, 128000, (void*)&Epos_interaction::ecatcheck, (void*) &ctime);
                                                                         ^
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp: In member function ‘void Epos_interaction::initialized_thread()’:
/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/src/final.cpp:595:73: warning: converting from ‘void (Epos_interaction::*)()’ to ‘void*’ [-Wpmf-conversions]
          osal_thread_create(&thread1, 128000, (void*)&Epos_interaction::ecatcheck, (void*) &ctime);
                                                                         ^
CMakeFiles/ethercat_soem.dir/src/final.cpp.o: In function `Epos_interaction::initialized_thread()':
final.cpp:(.text+0x5c13): undefined reference to `osal_thread_create'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/devel/.private/ethercat_soem/lib/ethercat_soem/ethercat_soem] Error 1
make[1]: *** [CMakeFiles/ethercat_soem.dir/all] Error 2
make: *** [all] Error 2
cd /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/build/ethercat_soem; catkin build --get-env ethercat_soem | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << ethercat_soem:make                   [ Exited with code 2 ]      
Failed    <<< ethercat_soem                        [ 4.7 seconds ]             
[build] Summary: 1 of 2 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 6.3 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.

mgruhler, 2019-08-02:
As discussed in https://github.com/mgruhler/soem/pull/24#issuecomment-515969618

> please include exactly which version you are using (from source, if yes, which branch/commit-sha)? from debs?) as well as the CMakeLists.txt (shortened to the relevant content OR link to repo) and the exact error message via copy'n'paste.

Thanks for the error message. Please format correctly in the future.

A guess: did you add `target_link_libraries(${PROJECT_NAME} soem)?

MRIAC, 2019-08-02:
below i am putting my CMakeLists.txt

cmake_minimum_required(VERSION 2.8.3)
project(ethercat_soem)

set(CMAKE_C_FLAGS "-std=c99" )
set(CMAKE_CXX_FLAGS "-fpermissive")

  find_package(catkin REQUIRED COMPONENTS message_generation std_msgs sensor_msgs roscpp soem)
  #find_package(catkin osal)

  # Declare that this catkin package's runtime dependencies
 catkin_package(
 INCLUDE_DIRS include 
 LIBRARIES ethercat_soem
 CATKIN_DEPENDS  roscpp soem osal oshw
 #CATKIN_DEPENDS sensor_msgs
 #DEPENDS lpthread 
)

include_directories(include
     include/ethercat_soem/
     ##include/ethercat/
     ${catkin_INCLUDE_DIRS}
     ${soem_INCLUDE_DIRS}/soem
    )

  #make the etehrcat_soem library
  add_library(etehrcat_soem
#	    src/Maxon/osal.cpp
#	    src/Maxon/oshw.cpp
#	    src/Maxon/nicdrv.cpp
#	    src/Maxon/ethercatsoe.cpp
#	    src/Maxon/ethercatprint.cpp
#	    src/Maxon/ethercatmain.cpp
#	    src/Maxon/ethercatfoe.cpp
#	    src/Maxon/ethercatdc.cpp
#	    src/Maxon/ethercatconfig.cpp
#	    src/Maxon/ethercatcoe.cpp
#	    src/Maxon/ethercatbase.cpp
	   src/final.cpp

)
 

add_executable(${PROJECT_NAME} src/final.cpp)

#target_link_libraries(soem ${catkin_LIBRARIES} soem)

add_dependencies(${PROJECT_NAME} ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
add_dependencies(${PROJECT_NAME} etehrcat_soem)

 target_link_libraries(${PROJECT_NAME} soem
   ${catkin_LIBRARIES}
 )

#add_dependencies(ethercat ${catkin_EXPORTED_TARGETS})
#target_link_libraries(ethercat
 #${catkin_LIBRARIES})

################################################################################
# Install
################################################################################
#install(TARGETS ethercat
 # ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
 # LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
 # RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
#)

#install(DIRECTORY include/${PROJECT_NAME}/
#  DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
#)

################################################################################
# Test
################################################################################

MRIAC, 2019-08-02:
@mgruhler  please get my CMakeLists.txt below.

cmake_minimum_required(VERSION 2.8.3)
project(ethercat_soem)

set(CMAKE_C_FLAGS "-std=c99" )
set(CMAKE_CXX_FLAGS "-fpermissive")

  find_package(catkin REQUIRED COMPONENTS message_generation std_msgs sensor_msgs roscpp soem)
 

  
 catkin_package(
 INCLUDE_DIRS include 
 LIBRARIES ethercat_soem
 CATKIN_DEPENDS  roscpp soem osal oshw
 
)

include_directories(include
     include/ethercat_soem/
     ${catkin_INCLUDE_DIRS}
     ${soem_INCLUDE_DIRS}/soem
    )

  add_library(etehrcat_soem
	   src/final.cpp

)
 

add_executable(${PROJECT_NAME} src/final.cpp)

add_dependencies(${PROJECT_NAME} ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
add_dependencies(${PROJECT_NAME} etehrcat_soem)

 target_link_libraries(${PROJECT_NAME} soem)

mgruhler, 2019-08-02:
You still haven't answered my question:

> please include exactly which version you are using (from source, if yes, which branch/commit-sha)? from debs?)

MRIAC, 2019-08-03:
I am using soem v1.3.0 and trying to link my own final.cpp ros publishers node to this.

mgruhler, 2019-08-03:
There are a few minor things that seem off with your `CMakeLists.txt`, but one is probably the issue you are facing.

You are building two targets

- an executable called like your project
- a library called `etehrcat_soem` --> note the typo, which is why Cmake is not complaining for a duplicate target name. 

You don't link anything against your library. So I guess this is where your problem is.

As the library is built from the same `final.cpp` as the executable, you should decide which one you want and remove the other.

MRIAC, 2019-08-05:
@mgruhler  thanks for reply, now i getting such error.

CMake Error at /home/atugv/Manpack_Drive_code/ethercat_motor/catkin_ws/src/ethercat-soem/CMakeLists.txt:50 (target_link_libraries):
  Cannot specify link libraries for target "ethercat_soem" which is not built
  by this project.

And this is my CMakeLists.txt file after your editing.
cmake_minimum_required(VERSION 2.8.3)
project(ethercat_soem)

set(CMAKE_C_FLAGS "-std=c99" )
set(CMAKE_CXX_FLAGS "-fpermissive")

  find_package(catkin REQUIRED COMPONENTS message_generation std_msgs sensor_msgs roscpp soem)
 

  
 catkin_package(
 INCLUDE_DIRS include 
 LIBRARIES soem
 CATKIN_DEPENDS  roscpp soem osal oshw
 
)

include_directories(include
     include/ethercat_soem/
     ${catkin_INCLUDE_DIRS}
     ${soem_INCLUDE_DIRS}/soem
    )

  add_library(etehrcat_soem
	   src/Maxon/osal.c
	   src/Maxon/oshw.c
	   src/Maxon/nicdrv.c
	   src/Maxon/ethercatsoe.c
	   src/Maxon/ethercatprint.c
	   src/Maxon/ethercatmain.c
	   src/Maxon/ethercatfoe.c
	   src/Maxon/ethercatdc.c
	   src/Maxon/ethercatconfig.c
	   src/Maxon/ethercatcoe.c
	   src/Maxon/ethercatbase.c

)
 

add_executable(ethercat src/final.cpp)

#add_dependencies(${PROJECT_NAME} ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
#add_dependencies(${PROJECT_NAME} soem)

 target_link_libraries(${PROJECT_NAME} soem)

install(TARGETS ethercat_soem
   ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
   LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
   RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
 )

install(DIRECTORY include/${PROJECT_NAME}/
  DESTINATION ${CATKIN_PACKAGE_INCLUDE_DESTINATION}
  FILES_MATCHING PATTERN "*.h"
)

mgruhler, 2019-08-05:
The error is right there:

You try to link against a target that does not exist.

You build two targets:

- `ethercat` in the `add_executable` call
- `etehrcat_soem` in the `add_library` call. Note *the typo etehrcat*

In the `target_link_libraries` call you link against `${PROJECT_NAME}`, which is `ethercat_soem` (no typo here).

I suggest you familiarize yourself a little bit more with the Cmake build system.

MRIAC, 2019-08-05:
@mgruhler please can you provide me any other link or reference to be follow..
But before that i want you to understand my requirements..i have written my own C++ ethercat ros node with taking help of simple_test.c and my drive is working fine with simple_test.c. But my concern is to write my own publisher node but my doubt is whether soem library which is written in C can be integrated with C++ ros node. i have tried a lot but unfortunately its not happening.

MRIAC, 2019-08-06:
@mgruhler thank you so much for your help and support..now i am able to link soem library to C++ code it was issue with CMake.Lists..

But can you suggest me how to run the motor code without root..??

TheEngineer96, 2019-08-06:
@MRIAC @mgruhler I have a similar problem (I think) with calling the C library from C++. I'm building a ROS-node that just grabs slave data over EtherCAT and publishes it on a topic. 

I first got the same error messages like MRIAC but following this thread has solved them. Now I keep getting following error messages:
```
/opt/ros/kinetic/include/soem/ethercatmain.h:490:33: error: ‘ec_mbxbuft’ was not declared in this scope
 int ec_mbxreceive(uint16 slave, ec_mbxbuft *mbx, int timeout);
                                 ^
```
This goes on for all the header files.

Is there some direction you could point me in to look at?

I'm working with ROS-Kinetic and Ubuntu 16.04 LTS with the SOEM v1.3.0 library.

mgruhler, 2019-08-06:
@MRIAC there is a tool called [ethercat_grant](https://github.com/shadow-robot/ethercat_grant) that helps you with that. I should really document that (#28).
Please close this issue if your questions are resolved.

@TheEngineer96 Can this be reproduced? This is very interesting. How can a `uint8` not be declared? I have no clue how upgrading from the debian package version to the version from #24 (I guess?) should help...
If this can be reproduced and should be investigated, please open a new issue with instructions how to reproduce.

TheEngineer96, 2019-08-07:
@mgruhler I'll have a look if I can reproduce the errors and come back to you.

MRIAC, 2019-08-08:
@mgruhler i am not able to run the node without root..i have followed your instruction (#28 ).
below is my launch file.

<launch>
<node name="final" pkg="ethercat" type="soem" output="screen" clear_params="true" respawn="true" launch-prefix="ethercat_grant" />
</launch>

**i have mentioned the error which i am getting after running the launch file.**

[final-2] process has died [pid 32563, exit code -6, cmd ethercat_grant /home/atugv/M1/catkin_ws/devel/lib/ethercat/soem __name:=final __log:=/home/atugv/.ros/log/ee205876-b9dc-11e9-9348-d8fc93b0a946/final-2.log].
log file: /home/atugv/.ros/log/ee205876-b9dc-11e9-9348-d8fc93b0a946/final-2*.log
[final-2] restarting process
process[final-2]: started with pid [32584]
[ INFO] [1565269456.459683255]: main is running

[ INFO] [1565269456.461595326]: Running EPOS INTERACTION
[ INFO] [1565269456.461798269]: Starting simple test

*** stack smashing detected ***: /home/atugv/M1/catkin_ws/devel/lib/ethercat/soem terminated

mgruhler, 2019-08-12:
@TheEngineer96 interesting. It isn't clear to me why this would be an issue. any follow-up would be appreciated.

@MRIAC you **are** able to run the node without root. Your log even states `main is running`. So all good there. You **are not**, however, having a working executable.
stack smashing can occur in buffer overruns when writing over the end of an array or trying to write to an uninitialized pointer. Check your arrays and pointers and make sure this is correct.

This is not a SOEM issue anymore, but something in your node implementation. Thus, I'm closing this issue as the original (and follow up) questions related to SOEM have been fixed.

closed 2019-08-12 by mgruhler