orocos/soem #31: SOEM to ease CI for ROS-i tests

calvama, 2019-10-08:
**Is your feature request related to a problem? Please describe.**
Using Travis (or some other platform-like) for Continuous Integration may require a physical robot for regression tests (i.e. robot drivers)

**Describe the solution you'd like**
I was imaging about a SOEM version for testing, so it should keep track on the calls and returning as if a physical robot were responding. Does this make sense?

**ROS Distro (please check as appropriate):**
 - [X] Kinetic
 - [ ] Lunar
 - [X] Melodic
 - [ ] Other: ...

**Hardware setup**
No hardware at all.

mgruhler, 2019-10-09:
> I was imaging about a SOEM version for testing, so it should keep track on the calls and returning as if a physical robot were responding. Does this make sense?

Yes, this makes sense.

What you are asking for is basically an "ethercat-emulator", that you can use to do regression tests with. So you can fire this up instead of actual hardware, send commands the "regular" ethercat way via `SOEM` and have the emulator react to it. Correct?

The thing is:
Each Ethercat Slave is a device which allows specific commands, and cannot deal with others. Even though standardized, like e.g. CoE, the available commands and error codes depend on the actual implementation of the slave. And not all slaves support all Ethercat Modes (CoE, EoE, ...).

So what do you expect of the "emulator"? Should it actually support the specific device with its specific characteristics? I.e. you can test if "that specific robot" works? Or should this be some general response, in the sense that "this command went through the master to the slave and came back"? But what would the latter actually provide?

So, I can think of two ways to achieve this:
1. Implement the "general response slave" (if this is possible). I'm not sure how this should be done, and what value this would provide. But this should then be targeted [upstream](https://github.com/OpenEtherCATsociety/SOEM) and not within the ROS wrapper package.
1. Implement a "software slave" that emulates your hardware. This is then specific to the device, and would require you to implement at least all the functionalities you want to test. But how do you test this piece of code, then? You could use the [Simple Open Ethercat Slave](https://github.com/OpenEtherCATsociety/SOES) to do this. But in the end, as the HW-Driver itself, this would be HW-specific and thus the right place is also not in this package.

If you feel there is something I missed, please let me know.
If not, I think this issue can be closed, as I don't see how this should be tackled in this repo...

calvama, 2019-10-09:
Let me try to summarize the scenario:

(Test case)>(Robot driver)>(SOEM)>(Physical Network)>(RobotHW)

So, when somebody is developing a driver, for sure there is access to the HW. So the requests and replies from the robot could be recorded as "particular calls and responses" for that particular test case.

So for each testcase a recorded file should be available. In regression testing on the cloud, this file would be used to verify that the driver is doing the proper calls (and would require simulated replies to continue).

Our first attempt for not modifiying SOEM, was to sniff directly on the wire, but there is quite a lot of ethercat traffic and would require a re-implementation of the protocol, since some data changes (as timestamps).

So we think it may be easier to create a record&playback mode in SOEM, which may somehow (not sure how to do this right now) be activated to be as more transparent as possible for the build process and without damaging the performance .

In this way a "general response software slave" could be achieved, and the buildchain would not be modified.

The Robot driver code could not be changed, otherwise it would be testing some other thing.

mgruhler, 2019-10-09:
Okay, this clarifies the scope.
So it would really be a regression test as in "this is what it was before and if nothing changes the messages then this will still work". However, this also means that for any change (maybe in timing, or if you only add one other configuration call in the driver) the recording would be useless. Correct?

There are quite a lot more of those questions coming to my mind. But I'll stop here.

This is something that should really be asked and targeted at the [SOEM upstream repo](https://github.com/OpenEtherCATsociety/SOEM). This is nothing that is specific to ROS but a general feature request. As such, this wrapper repo is not the right place to handle this.

I'll close this issue and suggest you ask this upstream (ideally link to this issue here). There are the people having the deep technical insight into SOEM that you need for such a feature.

If this is implemented and makes it into an upstream release, I'll happily incorporate it into this repo. I'll also support with incorporating this in a fork/dev branch, if this is needed.

If you feel that this is something that is specific to ROS and needs to be handled within the ROS driver, please feel free to reopen and explain why you think this is the case.

closed 2019-10-09 by mgruhler