orocos/soem #37: ROS Noetic release

Olavhaasie, 2020-05-18:
**Question**
Are you planning on making a release for ROS Noetic?

mgruhler, 2020-05-18:
Yes, a release would be great. I don't have a noetic machine at the moment, though. So I don't know when I'll get around to test that.

As this is pretty much cmake only, I guess this should be a piece of cake, though...

Olavhaasie, 2020-06-12:
I was able to successfully build the master branch on ROS Noetic without any changes. However, the [Noetic migration guide](https://wiki.ros.org/noetic/Migration#Increase_required_CMake_version_to_avoid_author_warning) suggest upgrading minimum CMake version to 3.0.2.

mgruhler, 2020-06-19:
@Olavhaasie thank you for your patience.

See ros/rosdistro#25540.

Please report back if you experience any issues.

closed 2020-06-19 by mgruhler