orocos/soem #38: SDO comunication problem

AlexCorrado, 2020-06-17:
Goodmorning to all,
I'm using a Pc with Ubuntu as Ethercat master and an ethercat slave which uses a LAN9252.
The slave is connected to a pc with window which is used to load the SOES program and once the software has been loaded, it remains connected in debug mode for checking the variables. I have a problem with the SDO communication. When i run, on my master, the slaveinfo.c from command window with the optional parameter -sdo i obtain the following results:

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp2s0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
Name: SSC_Device
Output size: 1024bits
Input size: 1024bits
State: 4
Delay: 0[ns]
Has DC: 1
DCParentport:0
Activeports:1.0.0.0
Configured address: 1001
Man: 00000afc ID: 00010000 Rev: 00000001
SM0 A:1000 L: 128 F:00010026 Type:1
SM1 A:1200 L: 128 F:00010022 Type:2
SM2 A:1400 L: 128 F:00010064 Type:3
SM3 A:1600 L: 128 F:00010020 Type:4
FMMU0 Ls:00000000 Ll: 128 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
FMMU1 Ls:00000080 Ll: 128 Lsb:0 Leb:7 Ps:1600 Psb:0 Ty:01 Act:01
FMMUfunc 0:1 1:2 2:0 3:0
MBX length wr: 128 rd: 128 MBX protocols : 04
CoE details: 0d FoE details: 00 EoE details: 00 SoE details: 00
Ebus current: 0[mA]
only LRD/LWR:0
ec_slave[cnt].mbx_proto: 04
printSDO: 1
End slaveinfo, close socket
End program

The problem is that i don't see the CoE object dictionary. I checked the exchange of packages between master and slave, i saw that the master sent a request to obtain the informations but my slave don't receive them. From the slave debugger I checked the SM_status (Sync manager status SM1 0x080d) that is the register that indicate if the mailmox is full and can be read and i saw that this register doesn't change value and then my slave doesn't pass into read SDO mode. Someone know what is the problem? the eeprom memory need to have a particular structure to allow a correct SDO comunication? Or maybe is only a slave software problem?

Thanks.

mgruhler, 2020-06-17:
Is this in any way related to the Robot Operating System ROS?

AlexCorrado, 2020-06-17:
Yes, I'm using SOEM in Ubuntu 16.04 ROS kinetic. I'm working with the SOEM package developed here

mgruhler, 2020-06-17:
Well, this package merely wraps [upstream SOEM](https://github.com/OpenEtherCATsociety/SOEM) for the use in ROS.

I don't see anything in your query that is directly ROS related. So it would be best to ask this at the upstream repository.
I'll be happy to help with any ROS related queries, though.

Please post a new issue at the upstream SOEM repo and link to this query.
Thank you.

mgruhler, 2020-06-19:
Seeing that this has been asked at upstream OpenEthercatSociety/SOEM#410, I'll close this issue.

closed 2020-06-19 by mgruhler