orocos/soem #40: Can't find headers when using SOEM from source

MatthijsBurgh, 2020-08-03:
**Describe the bug**
When using SOEM 1.4 from APT, I don't have any problems. But when using SOEM 1.4 from source, I get a missing headers error. I have tried finding SOEM directly or via catkin as component, both fail.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/tue-robotics/ethercat_interface.git in catkin workspace and checkout the `fix/ci_2020` branch
2. Clone soem in catkin workspace and checkout melodic branch
3. Install ` ros-ROS_DISTRO-hardware-interface ros-ROS_DISTRO-controller-manager ros-ROS_DISTRO-roscpp`
4. Compile with 'catkin build'
5. See error (https://travis-ci.com/github/tue-robotics/ethercat_interface/jobs/367803096#L745)
```
Errors     << ethercat_interface:make /home/amigo/ros/melodic/system/logs/ethercat_interface/build.make.000.log
In file included from /home/amigo/ros/melodic/system/src/ethercat_interface/src/drivers/./el4xxx.h:3:0,
                 from /home/amigo/ros/melodic/system/src/ethercat_interface/src/drivers/el4xxx.cpp:1:
/home/amigo/ros/melodic/system/src/ethercat_interface/include/ethercat_interface/driver.h:10:10: fatal error: soem/ethercattype.h: No such file or directory
 #include "soem/ethercattype.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/amigo/ros/melodic/system/src/ethercat_interface/src/drivers/./el5101.h:3:0,
                 from /home/amigo/ros/melodic/system/src/ethercat_interface/src/drivers/el5101.cpp:1:
/home/amigo/ros/melodic/system/src/ethercat_interface/include/ethercat_interface/driver.h:10:10: fatal error: soem/ethercattype.h: No such file or directory
 #include "soem/ethercattype.h"
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ethercat_interface.dir/src/drivers/el4xxx.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/ethercat_interface.dir/src/drivers/el5101.cpp.o] Error 1
make[1]: *** [CMakeFiles/ethercat_interface.dir/all] Error 2
make: *** [all] Error 2
cd /home/amigo/ros/melodic/system/build/ethercat_interface; catkin build --get-env ethercat_interface | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

**Expected behavior**
Compiling ;)

**ROS Distro (please check as appropriate):**
 - [ ] Kinetic
 - [ ] Lunar
 - [x] Melodic

**SOEM ROS Wrapper version (please check as appropriate):**
 - [ ] from apt: 1.3.0
 - [x] from apt: 1.4.0
 - [ ] from source: 1.3.0
 - [x] from source: 1.4.0
 - [ ] Commit SHA:

**`CMakeLists.txt`**
```
cmake_minimum_required(VERSION 2.8.3)

## Use C++ 11
add_compile_options(-std=c++11)

project(ethercat_interface)

## Find catkin and any catkin packages
find_package(catkin REQUIRED COMPONENTS
  roscpp
  soem
  hardware_interface
  controller_manager)

## Declare a catkin package
catkin_package(CATKIN_DEPENDS roscpp
               INCLUDE_DIRS include
               LIBRARIES ${PROJECT_NAME})

include_directories(
  include
  ${catkin_INCLUDE_DIRS}
)

# Ethercat interface library
add_library(${PROJECT_NAME}
    ${HEADERS}

    # Driver implementations
    src/drivers/el200x.h
    src/drivers/el4xxx.cpp
    src/drivers/el5101.cpp
    src/drivers/tuees030.cpp

    # IO implementations
    src/io/analog_input.h
    src/io/analog_output.h
    src/io/digital_input.h
    src/io/digital_output.h
    src/io/encoder.h

    src/interface.cpp
    src/driver.cpp
    src/io.cpp
    src/input.cpp
    src/output.cpp
)
target_link_libraries(${PROJECT_NAME} ${catkin_LIBRARIES})

# #### Test executables ####
# Ethercat interface node
add_executable(ethercat_interface_node test/ethercat_interface_node.cpp)
target_link_libraries(ethercat_interface_node ${PROJECT_NAME} ${catkin_LIBRARIES})

# List interfaces
add_executable(list_interfaces test/list_interfaces.cpp)
target_link_libraries(list_interfaces ${PROJECT_NAME} ${catkin_LIBRARIES})

# Test AI
add_executable(test_ai test/test_ai.cpp)
target_link_libraries(test_ai ${PROJECT_NAME} ${catkin_LIBRARIES})

# Test AO
add_executable(test_ao test/test_ao.cpp)
target_link_libraries(test_ao ${PROJECT_NAME} ${catkin_LIBRARIES})

# Test DO
add_executable(test_do test/test_do.cpp)
target_link_libraries(test_do ${PROJECT_NAME} ${catkin_LIBRARIES})

# Test encoder
add_executable(test_encoder test/test_encoder.cpp)
target_link_libraries(test_encoder ${PROJECT_NAME} ${catkin_LIBRARIES})
```

**Hardware setup**
Just compiling issues at the moment.

P.S.
You are doing a good job since you have taken over this repo.

MatthijsBurgh, 2020-08-03:
I think that the changes in `cmake/soem-ros-extras.cmake.in` in 9b40aa1fe336d974582868a7c7966cc9e03959ed should not have been done. Or the line without soem should have been kept." By removing the line without soem the headers can only by include without the soem prefix `#include <XXX.h>`. This not desired, because using the prefix makes clear from which package the headers are. But al least also #include <soem/XXX.h>` should be supported.

Why it does work with the system package and not when using source: `/opt/ros/ROS_DISTRO/include` is in `catkin_INCLUDE_DIRS`, therefore `#include <soem/XXX.h>` does work. `soem_INCLUDE_DIRS` is `/opt/ros/ROS_DISTRO/include/soem` and is ignored when using #include <soem/XXX.h>`. When using SOEM from source, there is no shared include folder as all packages have their include at `catkin_ws/devel/.private/PKG/include`. So that path needs to be in `soem_INCLUDE_DIRS`. But it is not. `catkin_ws/devel/.private/soem/include/soem` is and therefore the includes without prefix do work.

Most package provide an `_INCLUDE_DIRS` of the form: `/opt/ros/ROS_DISTR/include:/usr/include:/opt/ros/ROS_DISTRO/include/PKG`
SOEM should do the same,

mgruhler, 2020-08-03:
@MatthijsBurgh thanks for the bug report and the analysis. I agree. Reverting the changes to `cmake/soem-ros-extras.cmake.in` will fix it.

Sorry, I obviously didn't do enough testing on this. Actually, once you install the package provided by apt, it compiles (at least on my machine), even if you have it from source in your workspace. This is probably why I missed it. I have to purge the apt package to be able to reproduce the issue.

I'll try to push a fix and trigger a new release later today.

What do you mean with

> Most package provide an _INCLUDE_DIRS of the form: /opt/ros/ROS_DISTR/include:/usr/include:/opt/ros/ROS_DISTRO/include/PKG
SOEM should do the same,

?
As far as I see, reverting the change above should provide this again. Or am I missing something?

MatthijsBurgh, 2020-08-03:
Yes reverting solves is almost. I think `include` should come before `include/soem`

closed 2020-08-03 by mgruhler

mgruhler, 2020-08-03:
FYI, I just triggered a new release for melodic: ros/rosdistro#26026

commit 647adf35 (2020-08-03): fixes #40: export include directories properly
commit a4386b62 (2020-08-03): fixes #40: export include directories properly