orocos/soem #45: Send and Receive processdata Twice.

human2154, 2022-02-24:
**Question**

Hello. 
I'm a beginner in E-cat. And thank you for developing SOEM.

I use 'simpletest' for initializing slaves and a 1000Hz thread for calculating and sending-receiving E-cat data.
In a normal situation, the data is only sent at 1000ms timing. But in my graph, as the picture attached, it sends at 0ms and 1000ms twice.

I'll attach my code and timing graph. Please help me. Thank you.

[image]

`#include <ros/ros.h>

#include <stdio.h>
#include <string.h>
#include <math.h>
#include <iostream>
#include <list>
#include <signal.h>
#include <stdlib.h>

#include <sys/ioctl.h>
#include <sys/mman.h>
#include <sys/types.h>
#include <sys/resource.h>

#include <sys/sem.h>
#include <sys/shm.h>
#include <sys/ipc.h>

#include <cobalt/time.h>
#include <cobalt/unistd.h>
#include <cobalt/pthread.h>

#include <fcntl.h>   // File control definitions
#include <errno.h>   // Error number definitions
#include <termios.h> // POSIX terminal control definitions
#include "utility.h"
#include "ethercat.h"

using namespace std;
/*--------------------------Variables-----------------------------*/

#pragma region globalVars
// thread  EtherCAT  
#define EC_TIMEOUTMON 500
#define stack64k (64 * 1024)
#define NSEC_PER_SEC 1000000000
#define NUM_AXIS_TMP 1

pthread_t thread1; // ethercat
pthread_mutex_t mutex = PTHREAD_MUTEX_INITIALIZER;

char IOmap[4096] = {0}; // for XMC E-Cat
int expectedWKC;
boolean needlf;
volatile int wkc;
boolean inOP;
//---------------------------------------------------- 
#pragma pack(1)
typedef struct DSTRUCT_OUT     //  pdo
{                              // 1c12
    int32_t target_position;   // 0x607a
    int32_t target_velocity;   // 0x60ff
    int16_t max_torque;        // 0x6072
    uint16_t control_word;     // 0x6040 pdo domain 0x1604
    int16_t target_torque;     // 0x6071
    uint8_t mode_of_operation; // 0x6060
} out_ELMOt;
[... 442 lines cut ...]
        integral++;
    }
    if (delta < 0)
    {
        integral--;
    }
    *offsettime = -(delta / 100) - (integral / 20); // 100 20
    // gl_delta = delta;
}`

**ROS Distro (please check as appropriate):**
 - [O] Kinetic
 - [ ] Melodic
 - [ ] Noetic
 - [ ] Other: ...

**SOEM ROS Wrapper version (please check and complete as appropriate):**
 - [ ] from apt: 1.?.?
 - [O] from source: 1.?.?
 - [ ] Commit SHA:
 - sorry my package file was changed, so I can't find the version

**Hardware setup**
Describe exactly which type of hardware setup you have (i.e. which ethercat slaves)
- Numbers of ELMO twitter Platinum Motor Controller

human2154, 2022-02-27:
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on rteth0 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:ModuleSlotsDrive
 Output size: 48bits
 Input size: 48bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000009a ID: 01100002 Rev: 00120025
 SM0 A:1000 L: 256 F:00010026 Type:1
 SM1 A:1400 L: 256 F:00010022 Type:2
 SM2 A:1800 L:   6 F:00010064 Type:3
 SM3 A:1c00 L:   6 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   6 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000006 Ll:   6 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 256 rd: 256 MBX protocols : 0e
 CoE details: 2f FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
 CoE Object Description found, 492 entries.
 Index: 1000 Datatype: 0007 Objectcode: 07 Name: Device type
  Sub: 00 Datatype: 0007 Bitlength: 0020 Obj.access: 0007 Name: Device type
          Value :0x00020192 131474

**This is my slave info.**

 - First. when I do
`ec_slave[slc].blockLRW = 1;`
It can read and write PDOs but send twice

 - And is there any data about CoEdetails 8bit information???
 'ec_slave[slc].CoEdetails'
when I change this data 0x2F to 0x3F and turn off LRW, It communicates once but I cannot read and write PDOs.

mgruhler, 2022-02-28:
Hi @human2154 ,

Having a quick glance at your code, it seems you are calling `ec_send_processdata` twice in your `ecatthread` method. Once directly before the infinite while loop, and once at the end after the `clock_nanosleep` call.
So I would say that is expected behavior?

If not, please state more clearly what you are actually trying to do and what is called where. I'll not go over 14 pages of code without more information ;-)

About the general SOEM questions, I suggest to ask those at the [upstream repo](https://github.com/OpenEtherCATsociety/SOEM) where SOEM actually is developed. They don't seem to be ROS related. Maybe try to create a non-ROS minimal example. This should help with getting those resolved...

human2154, 2022-02-28:
First. Thank you for your reply.

 - In `ethercatthread` there are two `ec_send_processdata();`. The first one runs just once when initializing the thread. And in while loop, there's a second `ec_send_processdata();` function that runs repeatedly at 1kHz.

mgruhler, 2022-03-01:
Well with that bunch of code it is really hard to debug without having access to the hardware. Can you provide a minimal example (best, a single main function) that still shows the issue? With what program are you creating the graphs? Are you sure you are measuring correct?

I cannot help much with the issues your having and, again, IMO they are not ROS-related. So I suggest to use the original SOEM issue tracker and provide them with a non-ROS minimal example. They are so much more familiar with this than me. So they might be able to pinpoint the issue right away....

human2154, 2022-03-02:
Thank you.

In this program the ethercat thread below is the only read and write code.
The other code is just for initialize.
And I check this EtherCAT timing with the 'Hilsher net Analyzer tool' and it works well.
```
void *ecatthread(void *ptr)
{
    struct timespec ts, tleft;
    int ht;
    int cycletime;
    int toff = 0;
    struct sched_param p;
    p.sched_priority = sched_get_priority_max(SCHED_FIFO);
    pthread_setschedparam(pthread_self(), SCHED_FIFO, &p);
    clock_gettime(CLOCK_MONOTONIC, &ts);
    ht = (ts.tv_nsec / 1000000) + 1; /* round to nearest ms */
    ts.tv_nsec = ht * 1000000;
    cycletime = *(int *)ptr * 1000; /* cycletime in ns */
    printf("create ecatthread! \n");
    printf("pthread priority = %d\n", p.sched_priority);
    ec_send_processdata();
    //-----------------------------------------------------------------------------------------
    while (1)
    {
        /* calculate next cycle start */
        add_timespec(&ts, cycletime + toff);
        std::cout << cycletime << " cycletime | toff " << toff << std::endl;
        /* wait to cycle start */
        clock_nanosleep(CLOCK_MONOTONIC, TIMER_ABSTIME, &ts, &tleft);
        ec_receive_processdata(EC_TIMEOUTRET);
        if (ec_slave[0].hasdc)
        {
            ec_sync(ec_DCtime, cycletime, &toff);
        }
        ec_send_processdata();
    }
}
```

Thanks to recommend me to move this issue at the original SOEM git.
So, removing this issue is better for you? Or, just leave it?

human2154, 2022-03-02:
Thank you.

With your help, I solved some problems and write a new issue at original SOEM 
Here is the new link. Thank you.

https://github.com/OpenEtherCATsociety/SOEM/issues/597

closed 2022-03-03 by mgruhler