orocos/soem #46: Giving up maintainership - any volunteers?

mgruhler, 2022-03-01:
Hi all,

I'm finding less and less time (which has been very limited to begin with) mainting the ROS wrapper around the SOEM library, and thus, I'll not be able to support here much longer.
If anyone is interested, I'll be happy to transfer ownership of this repo to another individual or an org who will take over.

Mentioning a few of the people that have been active on issues / PRs around here: @smits @meyerj @etorta @ValATG @mictas @MatthijsBurgh @cplasberg @edhage

MatthijsBurgh, 2022-03-05:
Might be good to just transfer this to @orocos.

P.S. don't forget the `gbp` repo as well

mgruhler, 2022-03-07:
@MatthijsBurgh Thanks for the suggestion. I'd be happy to do that, given that the @orocos team agrees :-) How are you related to OROCOS? I see that you have orocos in your profile, I want to avoid putting it on people that don't know about this...

(Thanks for the pointer about the `gbp` repo, I've not forgotten about that :-) )

mgruhler, 2022-03-15:
Anyone?

@MatthijsBurgh I totally agree with your points above and that @orocos might be a suitable place. But I'm a bit reluctant to just transfer it without consent...

meyerj, 2022-03-15:
Apparently there was not a lot of activity here recently and the tracked version of SOEM is stable. So also not much maintenance work, except maybe for future ROS 2 releases... I see that a ROS 2 version was already proposed in https://github.com/mgruhler/soem/issues/20.

As one of the Orocos maintainers, I would agree with moving the repository to the @orocos organization. Still, anyone else who volunteers and wants to become the official maintainer for this package can get the necessary access rights on GitHub.

mgruhler, 2022-03-15:
@meyerj thanks for the offer. And yes, there has not much been going on, seeing that most stuff works and, as you say, the upstream repo is quite stable. Most traffic has been the occasional question about ehtercat/soem usage, but most is too specific for my limited knowledge and not ROS related, which is why I deflect most to the upstream repo.

I'll be happy to transfer it to the @orocos team. It is probably in better hands at an org than an individual anyways.

The biggest piece of work will definitely be a ROS2 "port". There are several available already, though non has been released, that last time I checked.

Thanks again, I'll transfer it in the next day or two and ping you again.

mgruhler, 2022-03-16:
@meyerj thanks for the invite. I've now transfered the soem repo to orocos. I assume you want me to transfer soem-gbp to the orocos-gbp group, but I've don't got the rights to do so. If you grant them to me, I'll move it. I've also added you as a collaborator. Maybe you can move it yourself (or maybe I have to grant you additional rights once you accept the invite).

Thanks!

meyerj, 2022-03-17:
> @meyerj thanks for the invite. I've now transfered the soem repo to orocos. I assume you want me to transfer soem-gbp to the orocos-gbp group, but I've don't got the rights to do so. If you grant [...]

That should work now, too. The checkboxes to allow team members to create new repositories were not enabled for the orocos-gbp organization.

I also accepted the invite, but I cannot move the repository out of your personal namespace, at least not with the current permissions.

mgruhler, 2022-03-18:
@meyerj Thank you. The Collaborator permissions remain a mistery to me :shrug:

I've now moved soem-gbp to the orocos-gbp org as well.

Thank you for your interest. This helps a lot. I think the orocos org is the right place for this repo. I've just recently lost access to my last Ethercat device, due to a "relocation" at work. So I just don't feel fit to maintain this any longer.

Thanks again!

closed 2022-03-18 by mgruhler